fprime-community/fprime-baremetal-reference #9: `The dependency target "Arduino_Drv_I2cDriver" of target   "BaremetalReference_Top" does not exist` while running `fprime-util generate`

capsulecorplab, 2023-07-23:
[image]

ethancheez, 2023-07-23:
@capsulecorplab 
This is because my PR for the I2C driver hasn't been merged yet. You can either switch to my `fprime-arduino` branch or manually add the I2C driver to your project. 

[https://github.com/ethancheez/fprime-arduino.git](https://github.com/ethancheez/fprime-arduino.git)

My main branch should be updated.

capsulecorplab, 2023-07-24:
Yup, adding your `fprime-arduino` library was able to generate the build cache! Ran into an undefined symbol error while attempting to build though
[image]

ethancheez, 2023-07-24:
> Yup, adding your `fprime-arduino` library was able to generate the build cache! Ran into an undefined symbol error while attempting to build though

I was unable to reproduce this error. Could you check your project if `lib/fprime-baremetal/Svc/TlmLinearChan` is included?

ethancheez, 2023-07-24:
> > > Yup, adding your `fprime-arduino` library was able to generate the build cache! Ran into an undefined symbol error while attempting to build though
> > 
> > 
> > I was unable to reproduce this error. Could you check your project if `lib/fprime-baremetal/Svc/TlmLinearChan` is included?
> 
> Yup, `lib/fprime-baremetal/Svc/TlmLinearChan` is there

This was my attempt to reproduce the error:
```
git clone --recurse-submodules  https://github.com/fprime-community/fprime-baremetal-reference.git
cd fprime-baremetal-reference/
fprime-util generate
fprime-util build -j4
```

The build succeeded, generating binaries for all deployments: `BaremetalReference`, `RadioPassthrough`, and `BaseDeployment`. My only theory on why the `undefined symbol` error occurs is if the `fprime-baremetal/Svc/TlmLinearChan/TlmLinearChan.fpp` file is not included.

If you navigate directly into the `lib/fprime-baremetal/Svc/TlmLinearChan` directory and run `fprime-util build`, does it build?

capsulecorplab, 2023-07-24:
> The `settings.ini` file should include:
> 
> ```
> library_locations:  ./lib/arduino/fprime-arduino:./lib/fprime-baremetal
> ```

Ah, that could be the issue. I commented that line out and replaced it with just a path to your fork of `fprime-arduino`
[image]

Let me fix that...

capsulecorplab, 2023-07-24:
I reverted any changes to settings.ini and set the head commit in `lib/arduino/fprime-arduino` to your fork and that fixed the build!
[image]
[image]

Now I just need to wait for my Teensy 4.1 to arrive in the mail to actually flash/test it :upside_down_face:

ethancheez, 2023-07-24:
> Now I just need to wait for my Teensy 4.1 to arrive in the mail to actually flash/test it :upside_down_face: 

The baremetal reference uses the RFM69 radio to communicate. You would also need two [RFM69 breakout boards](https://www.adafruit.com/product/3070?gclid=EAIaIQobChMI7J6N9qyogAMVcczCBB2KSAsWEAQYASABEgJnjPD_BwE) and two Teensy 4.1 boards. 

Or you can get two [Feather M0s](https://www.adafruit.com/product/3176)

capsulecorplab, 2023-07-24:
> The baremetal reference uses the RFM69 radio to communicate. You would also need two [RFM69 breakout [...]

Cool! Any advantage to getting two [Feather M0s](https://www.adafruit.com/product/3176) over two [RFM69 breakout boards](https://www.adafruit.com/product/3070?gclid=EAIaIQobChMI7J6N9qyogAMVcczCBB2KSAsWEAQYASABEgJnjPD_BwE)? What would the 2nd Teensy be for?

ethancheez, 2023-07-24:
> Cool! Any advantage to getting two [Feather M0s](https://www.adafruit.com/product/3176) over two [RFM69 breakout [...]

There are two separate deployments that communicate to each other via the radio: `BaremetalReference` and `RadioPassthrough`. 

One Teensy would have the BaremetalReference deployment, and the other Teensy would have the RadioPassthrough deployment. Your GDS would connect to the RadioPassthrough deployment to wirelessly command the BaremetalReference deployment.

capsulecorplab, 2023-07-24:
Ordered a couple of the radio breakout boards + an extra teensy - might as well test with all the platforms :slightly_smiling_face:

capsulecorplab, 2023-07-25:
I guess I'll go ahead and close this issue and open any new issues that may arise when attempting to connect and test with hardware. Thanks for all the help @ethancheez !

capsulecorplab, 2023-07-25:
I also just noticed there's a [433MHz](https://www.adafruit.com/product/3071) option that's license-free (which means I don't have to renew my ham license). Would this work as well?

ethancheez, 2023-07-25:
> I also just noticed there's a [433MHz](https://www.adafruit.com/product/3071) option that's license-free (which means I don't have to renew my ham license). Would this work as well?

Yes, I believe both the 433 MHz and 915 MHz bands do not require a HAM Radio License. As long as both of your radios operate at the same frequency, you should be fine!

capsulecorplab, 2023-08-18:
Re-opening this issue, as the `fprime-arduino` submodule doesn't contain @ethancheez's i2c drivers yet

closed 2023-08-18 by capsulecorplab