fprime-community/fprime-baremetal-reference #12: build error encountered with `build-fprime-automatic-teensy41/Components/Sensors/IMU_MPU9250/IMU_MPU9250ComponentAc.cpp`

capsulecorplab, 2023-08-18:
[image]
[image]

ethancheez, 2023-08-18:
Were there changes in `fprime` or `fpp-tools` @LeStarch? The [fprime commit](https://github.com/nasa/fprime/tree/92c9f0712358ee34789c69a4cdc013f17748ad86) currently checkout out in this submodule results in a successful build.

After I bumped up the version by running `git submodule update --remote --recursive` and rebuilding, I got the same error above. I reverted back `fprime` and I could successfully build again.

ethancheez, 2023-08-18:
After comparing the `IMU_MPU9250ComponentAc.cpp` files generated from both `fprime` versions, I found this discrepancy: 

The new cpp file is missing this:
```cpp
// ----------------------------------------------------------------------
  // Mutex functions for guarded ports
  // ----------------------------------------------------------------------

  void IMU_MPU9250ComponentBase ::
    lock()
  {
    this->m_guardedPortMutex.lock();
  }

  void IMU_MPU9250ComponentBase ::
    unLock()
  {
    this->m_guardedPortMutex.unLock();
  }
```

LeStarch, 2023-08-18:
It looks like these files changed with the change to `fpp-to-cpp`.

LeStarch, 2023-08-18:
@bocchino do you know any reason these two functions would not have been generated?

LeStarch, 2023-08-18:
@capsulecorplab @ethancheez did you purge and regenerate after bumping versions?  If not, this is likely the cause.

ethancheez, 2023-08-18:
> @capsulecorplab @ethancheez did you purge and regenerate after bumping versions?  If not, this is likely the cause.

Yes, I purged and regenerated

LeStarch, 2023-08-19:
@ethancheez I confirmed it is a bug in FPP.  I am filing an issue now: https://github.com/fprime-community/fpp/issues/315

capsulecorplab, 2023-08-19:
> @capsulecorplab @ethancheez did you purge and regenerate after bumping versions? If not, this is likely the cause.

Affirmative; Build cache was purged and regenerated

LeStarch, 2023-08-19:
Note: a PR has been submitted to fix this issue: https://github.com/fprime-community/fpp/pull/316

LeStarch, 2023-08-23:
The fix is merged into FPP! We'll get a new release out the door soon.