fprime-community/fprime-baremetal-reference #15: Flashing pre-compiled hex files

capsulecorplab, 2023-12-13:
For the Teensy boards, I'm wondering if it's possible to flash a previously compiled hex file without having to run `fprime-util build` to wait for the Teensy window to pop up

ethancheez, 2023-12-13:
You would still need the Teensyduino GUI to be open, but from there you can select any `.hex` file. If you save a `.hex` file under `build-artifacts/teensy41/BaremetalReference/bin/BaremetalReference.hex` outside of `build-artifacts`, you can re-select that file in the Teensyduino GUI and upload it.

capsulecorplab, 2023-12-13:
I think I figured it out! Just needed to dig around for where the Teensyduino GUI launcher was located
[image]

closed 2023-12-13 by capsulecorplab