fprime-community/fprime-baremetal-reference #19: `imu.PowerSwitch` state is set to OFF by default

capsulecorplab, 2024-02-07:
I'm not sure if this is intended behavior, but I had to send a `imu.PowerSwitch, ON` command in order to start receiving accelerometer and gyro telemetry. Is there a way to set the `imu.PowerSwitch` default value to ON?

ethancheez, 2024-02-08:
I had it `OFF` by default in case someone were to use the `BaremetalReference` without the IMU. If it was defaulted to `ON`, but the user did not have an IMU connected, there will be a continuous event in `fprime-gds` indicating that the I2C address cannot be found.

> Is there a way to set the `imu.PowerSwitch` default value to ON?

Yes, you can edit [this line](https://github.com/fprime-community/fprime-baremetal-reference/blob/main/Components/Sensors/IMU_MPU9250/IMU_MPU9250.cpp#L22) to:
```cpp
m_power(Fw::On::ON)
```

capsulecorplab, 2024-02-08:
Awesome! I'll close this issue as this is a feature not a bug, and that I can turn it off if need be. Thanks!

closed 2024-02-08 by capsulecorplab