fprime-community/fprime-baremetal-reference #20: Suggestions for maximizing signal range with RFM69 radio transceivers

capsulecorplab, 2024-02-09:
After having tested multiple radio configurations, the best range I seem to be able to get is ~10 ft between two Teensy 4.1 boards communicating via the 433 MHz Adafruit RFM69 transceiver radios (See https://github.com/mdrs-community/fprime-baremetal-reference/releases/tag/v3.4.0-0.1.1). Is there anything else I could perhaps tune to maximize my range? 10 ft seems awfully short for a couple of UHF transceivers.

ethancheez, 2024-02-09:
You can also try playing around with the power level. I think +20dBm is the max.

From:
```cpp
rfm69.setTxPower(14, true);
```

To:
```cpp
rfm69.setTxPower(20, true);
```

ethancheez, 2024-02-09:
Do you have antennas attached to your radios? Those make big differences.

With my RFM69HCW 915MHz using [these antennas](https://www.adafruit.com/product/4269?gad_source=1&gclid=CjwKCAiAlJKuBhAdEiwAnZb7lfs40-jjUGeIDG-HfkGSkbJ3uR-2FGpcOrhTv1rz0ontEtp8HXU5fxoC_U0QAvD_BwE), I was able to go as far as 50 feet with no communication issues.

And for one of my other satellite projects I worked on, if I remember correctly, these radios worked for hundreds of feet with much larger antennas (I think it was a Yagi antenna).

capsulecorplab, 2024-02-09:
> You can also try playing around with the power level. I think +20dBm is the max.
> 
> From:
> 
> ```c++
> rfm69.setTxPower(14, true);
> ```
> 
> To:
> 
> ```c++
> rfm69.setTxPower(20, true);
> ```

Just tried +20dB. Though, it doesn't seem to be performing all that much different than 14 :confused:

capsulecorplab, 2024-02-09:
> Do you have antennas attached to your radios? Those make big differences.

Yup, They're attached along the board [image]

capsulecorplab, 2024-02-09:
> And for one of my other satellite projects I worked on, if I remember correctly, these radios worked for hundreds of feet with much larger antennas (I think it was a Yagi antenna).
> 
> Edit: Our ground station had a Yagi antenna, the satellite had a much smaller copper antenna but better than the spring antenna.

Is there a recommended bill of materials for replicating such a Yagi antenna setup?

capsulecorplab, 2024-02-09:
I'm once again picking up more consistent telemetry downlinks by simply touching the RadioPassthrough antenna :smile:

ethancheez, 2024-02-09:
> Is there a recommended bill of materials for replicating such a Yagi antenna setup?

Hmm I don't have those (and I don't think I would be able to disclose that if I did). These were all custom fabricated.

capsulecorplab, 2024-02-09:
Hm, I guess I'll have to be the antenna itself for now :upside_down_face: