fprime-community/fprime-baremetal-reference #22: build breaks on latest Teensy `1.59.0` package index

capsulecorplab, 2024-04-09:
I've incurred multiple build failures before finally getting https://github.com/fprime-community/fprime-workspace-image/pull/53 merged - which included a CI build of (a semver-tagged release fork of) the [fprime-baremetal-reference](https://github.com/mdrs-community/fprime-baremetal-reference/releases/tag/v3.4.0-0.1.1). I discovered one of the culprits to be compiling with version `1.59.0` of the Arduino package index for Teensy. 

[image]

The fix that went into the aforementioned PR was to pin the Teensy package index to version `1.58.2`. If this approach is acceptable, I'm thinking the documentation in https://github.com/fprime-community/fprime-baremetal-reference/blob/main/docs/arduino-cli-install.md#setup-arduino-cli-for-teensy-adafruit-and-rpi-pico-w-boards should perhaps be updated to reflect that users should install the pinned version of the Teensy package index and/or add a reference to use the f[prime-workspace-image](https://github.com/fprime-community/fprime-workspace-image)

ethancheez, 2024-04-09:
I have also gotten the same issue with Teensy AVR v.1.59.0.

As of now, I am using forcing the use of v1.58.2 via:
```sh
arduino-cli core install teensy:avr@1.58.2
```

ethancheez, 2024-04-10:
It seems [Teensyduino 1.59](https://github.com/PaulStoffregen/cores/releases/tag/1.59) is now set to use C++17 standard. If I set the CMake standard to C++17 in `fprime-arduino/cmake/toolchain/teensy41.cmake`, all those errors seem to go away and the binary builds.

```cmake
set(CMAKE_CXX_STANDARD 17)
add_compile_options(-std=c++17 -std=gnu++17) # Force C++17 standard
```

This might be the way to go instead of forcing version 1.58.2, since there were a lot of changes in the Teensy core in 1.59.0.

ethancheez, 2024-04-21:
Set Teensyduino to use C++17 standard in [this commit](https://github.com/fprime-community/fprime-arduino/commit/07b261d8070fb2954e3a51d5df6617aa2666e8b2)

capsulecorplab, 2024-04-22:
> Set Teensyduino to use C++17 standard in [this commit](https://github.com/fprime-community/fprime-arduino/commit/07b261d8070fb2954e3a51d5df6617aa2666e8b2)

Triggered a CI build for the `fprime-workspace-image` that now also builds fprime projects (note-ably, the `fprime-baremetal-reference`) using https://github.com/fprime-community/fprime-arduino/commit/07b261d8070fb2954e3a51d5df6617aa2666e8b2 and Teensy package index version 1.59.0, but ran into a CMake error while attempting to generate the build cache with `fprime-util generate` (see https://github.com/fprime-community/fprime-workspace-image/actions/runs/8776253263/job/24079494214#step:4:1025). I'm guessing I have to update my CMake toolchain?

ethancheez, 2024-04-22:
Is your `fprime-baremetal-reference` using fprime v3.4.3? I currently have an open PR that updates the `main` branch of `fprime-baremetal-reference` to use fprime v3.4.3. The most recent `fprime-arduino` devel branch was to fix some changes after fprime v3.4.3.

capsulecorplab, 2024-04-22:
> Is your `fprime-baremetal-reference` using fprime v3.4.3? I currently have an open PR that updates the `main` branch of `fprime-baremetal-reference` to use fprime v3.4.3. The most recent `fprime-arduino` devel branch was to fix some changes after fprime v3.4.3.

Not yet. I'd have to remove the unit tests for the fprime-tutorial-math-component from the CI builds if I were to update to v3.4.3, since the [latest release only supports v3.4.0](https://github.com/fprime-community/fprime-tutorial-math-component/releases/tag/v3.4.0), which would be a breaking change, so I've been making do with v3.4.0 for now.