fprime-community/fprime-baremetal-reference #27: Cannot reproduce building reference for teensy41

ReggieMarr, 2024-11-01:
Hi I may be early to the party here but I forked this project to add support for the arduino due as part of a plan to bring up a reference application for a couple of the atsam uc's with freeRTOS + csp support.

My fork has fairly minimal changes really just adding some build scripts + a docker env and I actually removed my deployment in the latest but despite that attempting to run ```fprime-util generate teensy41 && fprime-util build teensy41``` I can't build the hex file and it fails with this output:

[teensyfailure.txt](https://github.com/user-attachments/files/17603020/teensyfailure.txt)

Is there something dumb I'm doing or is this expected for the current state of the repo ?

For reference [this is my fork](https://github.com/ReggieMarr/fprime-baremetal-reference/tree/due_deployment)

ethancheez, 2024-11-01:
Could you run `fprime-util version-check` and post the output here? Debugging your error now.

The current state of `fprime-baremetal-reference` should be working for the `teensy41` build.

ReggieMarr, 2024-11-01:
If you're trying to reproduce you can use the shell script to build the docker-image with ```./run.sh docker-build --clean``` and generating the build dir and then building it can be done with ```./run.sh build --clean```.

I also have a command for interacting with the container interactively and I used this to generate the version info like so:

```shell
fprime-baremetal-reference on  due_deployment:main [$!?⇡] via △ v3.22.1 took 12s 
❯ ./run.sh inspect fsw
Running fsw container interactively...
docker compose run --rm --no-deps --user 1000:1000 fsw bash
user@reggiemarr-t15p:/fsw$ fprime-util version-check
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.0-124-generic-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 24.3.1
Pip packages:
    fprime-tools==3.5.0
    fprime-gds==3.5.0
    fprime-fpp-*==2.2.0
```

ethancheez, 2024-11-01:
I have an arduino due myself, maybe I can attempt getting it built for the due after we figure out the teensy error :)

ethancheez, 2024-11-01:
only cmake versions >`3.26.0` will work. Try install:
```sh
pip install cmake==3.26
```

ReggieMarr, 2024-11-01:
That'd be great! I had previously leverage the [arduino-cmake-toolchain](https://github.com/ReggieMarr/Arduino-CMake-Toolchain) project to create a [CSP](https://libcsp.github.io/libcsp/) server on the arduino due using freeRTOS (in part because the chip it uses is pretty similar to a [rad tolerant variant](https://www.microchip.com/en-us/product/SAM3X8ERT).

closed 2024-11-01 by ReggieMarr

ethancheez, 2024-11-01:
> Btw would the containerization of deps be welcomed in this MR or is there another philosophy applied here ?

Lets hold off on the containerization specifically for `fprime-baremetal-reference`, all the tutorials utilize the python virtual environments and would like to keep things consistent.

ethancheez, 2024-11-01:
Also, I just update the Arduino cookiecutter to support v3.5.0. I see in your fork that you created a new deployment, but for prior F Prime versions.

ethancheez, 2024-11-01:
@ReggieMarr You can use [this](https://github.com/fprime-community/fprime-arduino/tree/add/arduinodue) as a starting point. I got an LedBlinker deployment working on my DUE, with fprime-gds commanding it. 

But had to do disable some components in F' due to 32-bit compatibility issues (and other random things I didn't look too deep into fixing):
```cmake
set_target_properties(Svc_FatalHandler PROPERTIES EXCLUDE_FROM_ALL TRUE)
set_target_properties(Svc_SystemResources PROPERTIES EXCLUDE_FROM_ALL TRUE) #U64
set_target_properties(Svc_Version PROPERTIES EXCLUDE_FROM_ALL TRUE)
set_target_properties(Svc_DpCatalog PROPERTIES EXCLUDE_FROM_ALL TRUE) # u64
set_target_properties(Svc_DpWriter PROPERTIES EXCLUDE_FROM_ALL TRUE) #u64
set_target_properties(Svc_DpManager PROPERTIES EXCLUDE_FROM_ALL TRUE) #64
set_target_properties(Svc_FileManager PROPERTIES EXCLUDE_FROM_ALL TRUE) #u64
set_target_properties(Svc_SeqDispatcher PROPERTIES EXCLUDE_FROM_ALL TRUE)
```

ReggieMarr, 2024-11-02:
I see, I'll probably maintain the container on my fork then imho this instance is sort of an example of why trying to use python virtual envs as a containerization strategy doesn't quiet cut the mustard.

That's awesome though thank you ! 

I found that for my initial csp server I had to upgrade the compiler version as arduino due uses one that's so old a lot of c++11 stuff doesn't work so that doesn't suprise me but also that's a great example of how to customize a build for a particular platform